pulkin/micropython #4: ADC

pulkin, 2019-10-29:
Analog-digital converter in `machine.ADC`

arlucio, 2019-11-22:
I've ported ADS1x15 Micropython lib for A9 (https://github.com/arlucio/ads1x15_A9)

As A9 has only 3 ADCs, it is pretty useful to expand ADCs number using the I2C interfaces. Meanwhile we don't port A9 ADC, it's also a good solution. Been using it without any problems.

Maybe it's a good idea to link it somewhere on the readme.

pulkin, 2019-11-25:
Awesome, people from the upstream might be interested putting it somewhere into `drivers` such that other ports will also benefit from your implementation. Publishing it via `pip` is another option.

pulkin, 2019-11-29:
I ported ADC for those two pins. Note that they are measuring up to 1V.

commit c5fb57ae (2019-11-29): Closes #4
closed 2019-11-29 by pulkin