pulkin/micropython #9: Move the `i2c` module to `machine`

pulkin, 2019-10-29:
To conform with the [API standard](https://docs.micropython.org/en/latest/library/machine.I2C.html).

garudaonekh, 2021-09-01:
hi, any update on this? I do not know how to migrate the available I2C libraries to this. many i2c libraries are based on the standard machine.I2C

pulkin, 2021-09-01:
You could look how it is implemented here for example.

https://github.com/pulkin/micropython/blob/master/ports/esp8266/machine_hspi.c

garudaonekh, 2021-09-02:
> You could look how it is implemented here for example.
> 
> https://github.com/pulkin/micropython/blob/master/ports/esp8266/machine_hspi.c

I think it is simpler just to write a python class that wraps your existing one so that user can jus