pulkin/micropython #20: 1-Wire/Open Drain support

arlucio, 2019-11-22:
I'm trying to enable 1-Wire support on A9. For that it's necessary to implement Open-Drain GPIO Output somehow by software using SDK GPIO lib, doing something like this: https://embeddedartistry.com/blog/2018/06/11/simulating-open-drain-gpio-in-software/

From what I see, would be necessary to create a way to define a Pin in machine_pin as OpenDrain, to be able to call pin.init(pin.OPEN_DRAIN, pin.PULL_UP) on drivers/onewire/onewire.py. I've stumbled in how to do that, will try further but any help would be appreciated. 

ESP8266/ESP32/STM32 are some ports that already support 1-Wire/OpenDrain.

huangguangw, 2020-03-29:
请问onewire在A9G上实现了吗？

arlucio, 2020-04-21:
No, for this is still necessary to implement software Open-Drain for it.

ahmadmicro, 2021-01-24:
any progress on this?