pulkin/micropython #26: MPY Example Code

sebi5361, 2020-01-25:
Thanks for your great work!

Do you have any Micropython example code to share to simplify my understanding of your different APIs?

I would like to transmit regularly some GPS (together with some AGPS) coordinates to a MQTT broker using SSL/TLS and the cellular connection.

Maybe you already wrote such piece of software?

pulkin, 2020-01-25:
Yep, some examples needed. Here you go with GPS over MQTT:
```python
# Get online
import cellular
cellular.gprs("internet", "", "")

# Import mqtt (download client if necessary)
try:
    from umqtt import simple
except ImportError:
    import upip
    upip.install("micropython-umqtt.simple")
    from umqtt import simple

# Turn GPS on
import gps
gps.on()

# Report location
name = "a9g-micropython-board"
server = "test.mosquitto.org"
topic = "a9g-micropython-board-topic"
print("To track messages run, for example\n  mosquitto_sub -h {server} -t \"{topic}\" -v".format(server=server, topic=topic))
import json
import time
client = simple.MQTTClient(name, server)
client.connect()
while True:
    data = json.dumps(gps.get_last_location())
    print("Publish", data)
    client.publish(topic, data)
    time.sleep(5)
```
Please note that I am going to swap longitude and latitude output in the nearest future.

pulkin, 2020-01-25:
Btw, please share examples you'd like to see here.

sebi5361, 2020-01-25:
Very nice. Thank you!

sebi5361, 2020-01-26:
Your examples are neat!
Could you add an example on how to connect to a MQTT server using credentials and ssl/tls?
Can't wait to receive my board.

pulkin, 2020-01-26:
Try replacing
```python
client = simple.MQTTClient(name, server)
```
with
```python
client = simple.MQTTClient(name, server, user="user", password="pass", ssl=True)
```
Please note that `ssl` is much slower (takes around 5s for a handshake). In addition, I did not test certificates (without supplying certificates your cell provider will be able to perform MITM). Maybe I need to merge upstream to make it more secure.

sebi5361, 2020-01-26:
> In addition, I did not test certificates (without supplying certificates your cell provider will be able to perform MITM).

Is the process as simple to use certificates? Something like:
```python
...MQTTClient(name, server, user="user", password="pass", ssl=True, certificate="certificate")
```

> Maybe I need to merge upstream to make it more secure.

I have noticed that you forked your project from _Neutree/micropython_ rather than _micropython/micropython_ so it might not be that trivial, but indeed merging upstream would be a must!

pulkin, 2020-01-26:
Please look here for available options:

https://github.com/micropython/micropython-lib/blob/master/umqtt.simple/umqtt/simple.py

sebi5361, 2020-01-26:
Based on your link and learning more about the [_ussl_](https://docs.micropython.org/en/latest/library/ussl.html) library, it seems that whether certificates work or not depends on the port.
I don't know on which port is based the _A9G_ port, but according to this [thread](https://github.com/micropython/micropython/issues/2781), it seems that, if it is based on the _ESP8266_ port, it should work with a recent build. That's a good news!

closed 2020-02-07 by sebi5361