pulkin/micropython #27: Cannot connect to the REPL

sebi5361, 2020-02-08:
I think I successfully built the firmware despite getting some warnings:
```bash
../../py/nlrsetjmp.c: In function ‘nlr_jump’:
../../py/nlrsetjmp.c:41: warning: ‘noreturn’ function does return

../../py/modbuiltins.c: In function ‘mp_builtin_round’:../../py/modbuiltins.c:497: warning: declaration of ‘floor’ shadows a global declaration
/home/vagrant/A9G/micropython/lib/GPRS_C_SDK/include/std_inc/math.h:328: warning: shadowed declaration is here

../../py/modmath.c: In function ‘mp_math_frexp’:../../py/modmath.c:238: warning: declaration of ‘significand’ shadows a global declaration
/home/vagrant/A9G/micropython/lib/GPRS_C_SDK/include/std_inc/math.h:397: warning: shadowed declaration is here

main.c: In function ‘MicroPyTask’:main.c:161: warning: unused variable ‘event’

modcellular.c:636: warning: ‘modcellular_sms_from_raw’ defined but not usedCC modgps.c

AS gchelper.sCC ../../lib/axtls/ssl/asn1.c

../../lib/axtls/ssl/tls1.c: In function ‘add_hmac_digest’:../../lib/axtls/ssl/tls1.c:695: warning: incompatible implicit declaration of built-in function ‘alloca’
../../lib/axtls/ssl/tls1.c: In function ‘send_packet’:
../../lib/axtls/ssl/tls1.c:1177: warning: incompatible implicit declaration of built-in function ‘alloca’

../../lib/axtls/crypto/rsa.c: In function ‘RSA_decrypt’:../../lib/axtls/crypto/rsa.c:150: warning: incompatible implicit declaration of built-in function ‘alloca’
```
Indeed, in the end, I got `BUILD OK` and the two `.lod` files got available in the `hex` folder.
Than, I believe I successfully burned both `.lod` files using `coolwatcher` as I didn't observe any error message.

But now, I cannot connect to the MicroPython REPL.
I only observe garbage in the console when running the `miniterm.py /dev/ttyUSB1 115200 --raw` command and pressing on the board reset button.

What could be my issue?
Should I connect the serial adapter to other pins than `HST_TX` and `HST_RX` to access the REPL?
Thks,

sebi5361, 2020-02-08:
I just found the answer to my problem:
The serial adapter needs to be connected to `TX1` and `RX1`.
Am closing this issue.

closed 2020-02-08 by sebi5361

pulkin, 2020-02-08:
I think it should be emphasized somehow.