pulkin/micropython #31: How to increase the size of the RX buffer?

sebi5361, 2020-02-17:
Dear Pulkin,
I am used to using an extension of Microsoft Visual Studio Code called PyMakr to develop MicroPython code more easily. It works fine on several boards such as the ESP8266.
There is a command to copy a .py file onto the board similarly to ampy.
But it doesn't work with the A9G board.
I believe it has to do with the A9G RX buffer being too small to store a big chunk of data.
If so, increasing the size of that buffer would solve my issue.
Could you tell me where to size that buffer in the code before building the A9G port?
Thanks,

pulkin, 2020-02-17:
Its 256 bytes in `uart.h` (same as esp8266 in `esp_mphal.c`). So the problem is likely to be elsewhere. Please specify exactly what do you mean by "it doesn't work". Does the file become broken?

pulkin, 2020-02-17:
Regarding your title question:
```
import machine
uart = machine.UART(0, rxbuf=512)
os.dupterm(None, 1)
os.dupterm(uart, 1)
```

sebi5361, 2020-02-17:
Indeed, it didn't solve my issue...
The error messages when uploading a project are quite laconic:
```python
>>>
Uploading project (main folder)...
Not safe booting, disabled in settings

Uploading to /...
Reading file status
Failed to read project status, uploading all files
Creating dir code
Creating dir lib
Creating dir lib/umqtt
[1/85] Writing file agps.py (4kb)
Failed to write file, trying again...
Failed to write file, trying again...
timeout
[2/85] Writing file boot.py (1kb)
Failed to write file, trying again...
Failed to write file, trying again...
timeout
...
```
So I decided to look at what is happening under the hood by sniffing at the TX and RX lines.
On the TX line:
```python
.
import ubinascii
f = open('project.pymakr', 'wb')
f.write(ubinascii.a2b_base64('W1siY29kZSIsImQiLCI1Njk0ZDA4YTJlNTNmZmNhZTBjMzEwM2U1YWQ2ZjYwNzZhYmQ5NjBlYjFmOGE1NjU3NzA0MGJjMTAyOGY3MDJiIl0sWyJsaWIiLCJkIiwiNzZiNWEzNTczOTEyNzZiMjgyYTUxNmY1NGY0OGVmM2MyMDdmNDZkODE5MmRjNThjMjA4ZDUxODNkMzg0MTVmOCJdLFsibGliL2xvZ2dpbmciLCJkIiwiZTI0MGY0MWY0N2QwMjM1MjE0MWU2NTJhYmRiNjQyMTVhOGE5NGRkNzYzMWE3ZTFhODRmNWFjZTFhZThhODEzNiJdLFsibGliL3Rpbnlsb3JhIiwiZCIsImI1ZWJhN2M5MzQwZWJlOGI0OWZhZTMzOGYyM2QwNjdlOTliY2FkMjVlZjE5ODRjZTRiMWFiYWJiYTc3YzhiM2MiXSxbImxpYi91YXN5bmNpbyIsImQiLCI1YTgyMjIyOTg3ZjJjNDE5YTkzY2RhY2JlY2YyMWRiZjgxYzhjNDc3OTAzZmIyNjBiYTI1NzdkZGM4MTc3OTFmIl0sWyJsaWIvdW1xdHQiLCJkIiwiM2U2MmQyYTQ0MjA1MDc0YWJkNTc3NzU5NGZlMTBjNDNjMzkyN2JmMWRkMjI1NWZmNjM2ZDk0MWVlZTk4M2QwMiJdLFsidW0='))
.
```
where dots . represent special characters like Ctrl+A for entering the Raw-REPL.
On the RX line errors are of these types:
```python
raw REPL; CTRL-B to exit
>OK'/'
>OKTraceback (most recent call last):
  File "<stdin>", line 3, in <module>
OSError: [Errno 5] EIO
>OKTraceback (most recent call last):
  File "<stdin>", line 3, in <module>
OSError: [Errno 17] EEXIST
>OKTraceback (most recent call last):
  File "<stdin>", line 4
SyntaxError: invalid syntax
```
Do you have any clue why those errors occur with the A9G board, whereas the process works fine on an ESP8266 board?

sebi5361, 2020-02-17:
By increasing the size of the RX buffer to 1024 I had more success.
Now the main error I see when sniffing the RX line is:
```python
>OKTraceback (most recent call last):
  File "<stdin>", line 2, in <module>
ImportError: no module named 'uhashlib'
```
So I tried to include the uhashlib library into the build by setting it on in mpconfigport.h by writing:
```c
#define MICROPY_PY_UHASHLIB                 (1)
#define MICROPY_PY_UHASHLIB_SHA1            (1)
#define MICROPY_PY_UHASHLIB_SHA256          (1)
```
but this resulted in errors during the compilation...
Do you know how to fix those compilation errors?

pulkin, 2020-02-17:
I see. I think `uhashlib` is already there but it cannot be enabled without fixing `ussl` issues. I opened #32 for that. No, I do not know how to fix compilation errors at this point.

sebi5361, 2020-02-17:
Thank you. I will use ampy for the moment.
Are you planing to update your fork with the latest MicroPython build?
Can I contribute to your port by completing your .md files with some extra information to help newcomers?

pulkin, 2020-02-17:
Yes, feel free to create a new issue about documentation. We certainly need
- general information (what the module and mpy can be used for);
- tutorials with some breadboard photos;
- detailed description of [examples](examples/) (we need more example ideas as well).

I see it as a separate `doc` folder mapped to github pages.

closed 2020-02-18 by sebi5361

sebi5361, 2020-02-20:
@pulkin: Despite having `hashlib` working properly now and setting the RX buffer to 1024B, some large files (4KB and above) wont copy onto the board.
Thus I believe the way the UART works on the A9G board is a bit different than with an ESP8266 where this issue doesn't occur. Strange~~
Fortunately for me, files larger than 4KB are quite rare and I can live without it.

pulkin, 2020-02-20:
I will investigate it once I have access to the module.