pulkin/micropython #32: Fix `ussl` and `uhashlib`

pulkin, 2020-02-17:
Resolve issues with the build system and move `MICROPY_*` variables from the `Makefile` back to `mpconfigport.h`. Enable `uhashlib` which causes build failures at this point.

pulkin, 2020-02-19:
The problem might be in `extmod/moduhashlib.c` which does not pick `MICROPY_SSL_*` variables and re-compiles stuff causing duplicate symbols.

pulkin, 2020-02-20:
@sebi5361, please check whether `ssl`, `hashlib`, `cryptolib` work for you. You may build from `uhashlib` or just download [these builds](https://dev.azure.com/gpulkin/micropython/_build/results?buildId=142&view=artifacts&type=publishedArtifacts).

sebi5361, 2020-02-20:
I was able to `import ssl`, `hashlib`, and `cryptolib` properly.
I was able to synchronise my local project folder with the board (for the most part) which involves using `hashlib`.
Thank you!

sebi5361, 2020-02-22:
Dear Pulkin,
Soon, will you merge branches *binary-patches* and *uhashlib* so I can benefit from both fixes together?

pulkin, 2020-02-22:
Sure, I just need to test both on my hardware.

pulkin, 2020-02-23:
Merged.

closed 2020-02-23 by pulkin