pulkin/micropython #34: UART performance

pulkin, 2020-02-23:
Make this work:
- all `ampy` commands
- uploading large chunks of data
- vscode plugin

sebi5361, 2020-02-24:
I have just tried with my setup (VSCode + PyMakr) with a project that contains big files.
Am still having some errors:
```python
Creating dir lib/tinylora
Creating dir lib/uasyncio
[1/27] Writing file boot.py (1kb)
[2/27] Writing file code/createstubs.py (14kb)
[3/27] Writing file code/credentials.py (0kb)
[4/27] Writing file code/stubber.py (14kb)
[5/27] Writing file lib/bme280.py (9kb)
Failed to write file, trying again...
Failed to write file, trying again...
Filecheck failed
[6/27] Writing file lib/core.py (10kb)
Failed to write file, trying again...
Failed to write file, trying again...
Failed to write file: [Errno 5] EIO
[7/27] Writing file lib/HerkuleX.py (11kb)
Failed to write file, trying again...
Failed to write file, trying again...
Failed to write file: [Errno 5] EIO
[8/27] Writing file lib/ina219.py (16kb)
Failed to write file, trying again...
Failed to write file, trying again...
Failed to write file: [Errno 5] EIO
[9/27] Writing file lib/pye.py (40kb)
Failed to write file, trying again...
Failed to write file, trying again...
Filecheck failed
[10/27] Writing file lib/qmc5883l.py (10kb)
Failed to write file, trying again...
Failed to write file, trying again...
Upload failed.: Filecheck failed: OSError: [Errno 5] EIO
 Please reboot your device manually.
```

sebi5361, 2020-02-24:
I have just tried again after having erased completely all the files from the board.
And it worked perfectly with my setup. Files up to 40KB were copied flawlessly.
Congratulations!

pulkin, 2020-02-24:
I fixed error codes: now it should display EIO 28 in your case indicating the lack of space.

sebi5361, 2020-02-25:
I tried to observe this new EIO 28 error code but I didn't manage to make my board bump into such an error yet.

arlucio, 2020-02-27:
I saw the make "all ampy commands" to work and decided to post this here.

I've updated to the last version of the fw and I couldn't make _rmdir_ ampy command to work, _rm_ is working fine.

I was able to delete directories using _rshell_ and the _rm -r "DIR"_ command, and files using _rm -f "FILE"_

Tried _mpfshell_ and the _ls_ command don't work at all and crashes the connection, but I was able to delete files and directories using the  _rm_ command.

In all three options, connection was quite unrealible, many times I try to delete something and it just don't work at all, so I had to disconnect everything (power and communication port) and it started working again (lost quite some time cause I was just disconnecting the power or resetting module).

pulkin, 2020-02-27:
I hope that automated testing will take care of most of these issues. I am starting looking into it.

arlucio, 2020-02-28:
Ok, so I've installed the uhashlib fw build and now is the opposite, ampy and rshell doesn't work but mpfshell is working perfectly!

pulkin, 2020-02-28:
That does not clarify the issue. If you want to help here you may intercept and investigate the UART traffic using `interceptty /dev/ttyUSB0 bridge` for example.