pulkin/micropython #35: API evolution

sebi5361, 2020-02-27:
Dear pukin, here are some thoughts:
- Rather than giving the latest available location maybe `gps.get_location()` could yield `(None, None)` if not capable of getting the current location?
- Maybe `cellular.gprs(...)` would rather not throw an error if already connected as everything is fine,
- Could you provide the bit-wise specs of `get_network_status()`?
- Do you still plan to swap latitude and longitude?

pulkin, 2020-02-27:
- I think about giving access to the timestamp of the last reported location (if there is any). Unfortunately, GPS does not really provide something on demand: it just spams UART2 with its messages and the simplest thing one could do is to just save the last reported one.
- yep, we need some sort of `is_gprs`. It might be simpler to just have that `get_network_status` and some masking constants.
- I ended up with `longitude`, `latitude` but yes, it is probably better to swap them.

sebi5361, 2020-02-28:
Currently, is there an easy way to know if the given location is the current one? For example when the given gps time is actually accurate does it mean that the location is correct too?

pulkin, 2020-02-28:
I think yes but I am not really an expert here. Both time and location are from NMEA RMC sentence which probably means that they correspond to each other.

sebi5361, 2020-02-28:
Oh OK.
I modified the `sys.path` order to `['/lib', '/', '']` to `import cellular` from a *cellular.py* local file rather than the OS embedded version but it seems this doesn't work. Is this behavior normal?

pulkin, 2020-02-28:
I would assume yes. You could try hacking it through `sys.modules` as described [here](https://stackoverflow.com/questions/24948751/how-can-i-overload-a-built-in-module-in-python).

sebi5361, 2020-02-28:
I will try.
But according to [this post](https://forum.micropython.org/viewtopic.php?t=4622#p26728) regarding import paths:

> MicroPython uses `sys.path` to control which directories and the order that they are searched. The empty string represents frozen modules

I remember this used to work on the ESP8266.

pulkin, 2020-02-28:
All built-in modules are already loaded at the point you import them. I am not sure how to adjust it.

sebi5361, 2020-02-29:
Interesting. Didn't know. I wonder how this is made different with the other ports.
Anyhow that is not very important. I just wanted to overload the current modules with some small tweaks (such as swapping latitude and longitude) till the final adjustments arrive.

pulkin, 2020-03-01:
Swapped lat <-> lon with e4402b3dfb04153ae78611aa0a2d851db16fbca8. Same in [micropython-agps](https://github.com/pulkin/mpy-agps) with 4530ebff359c5fd0d2f1b5b9c522648bf3d111a6.

sebi5361, 2020-03-02:
Perfect! Shouldn't you swap lat and long in bs-dl.py as well?
`python -m bs-dl --mcc 208 --circle=LATITUDE,LONGITUDE,RADIUS -v --token TOKEN`

I have generated a bin file using bl-dl.py but I don't obtain the exact same results when running:
`coordinates = agps.get_location_local(cellular.agps_station_data(), "208.bin")` <- further away
`coordinates = agps.get_location_opencellid(cellular.agps_station_data(), token)` <- more precise

Could you tell me more about what bs-dl.py parameters to set to obtain the very same results?

sebi5361, 2020-03-02:
Additionally can you tell me if it is normal to get:
```
>>> cellular.get_imei()
'000000000000000'
```

pulkin, 2020-03-04:
> Perfect! Shouldn't you swap lat and long in bs-dl.py as well?

yes

> Could you tell me more about what bs-dl.py parameters to set to obtain the very same results?

I do not know. You get coordinates of base stations and signal levels but then you'll have to average them somehow. Probably I was too smart with averaging. It is less likely that base station data was updated. I will make a change for it to return the original data such that you can play with various averages to see what works best for you.

> Additionally can you tell me if it is normal to get ...

Well, somebody was mentioning that these modules do not really have a good IMEI.

sebi5361, 2020-03-04:
> I will make a change for it to return the original data such that you can play with various averages to see what works best for you.

I will try but sounds complex to me.

I have noticed that when rebooting the board with Ctrl-D the watchdog remains on (if set before).
Maybe this could be fixed~~

Could you provide an example on how to automatically connect to the APN when booting?
I implemented a loop but often it just won't connect forever.

pulkin, 2020-03-04:
> I have noticed that when rebooting the board with Ctrl-D the watchdog remains on (if set before).
Maybe this could be fixed

Is it like that for ESP? Then, yes: please open a new issue.

> Could you provide an example on how to automatically connect to the APN when booting?

That's also a reasonable request -> New Issue.

sebi5361, 2020-03-04:
OK will do.
I understand now that you prefer multiple single objective issues.
Will close this one.

closed 2020-03-04 by sebi5361

pulkin, 2020-04-22:
> Perfect! Shouldn't you swap lat and long in bs-dl.py as well?
> `python -m bs-dl --mcc 208 --circle=LATITUDE,LONGITUDE,RADIUS -v --token TOKEN`

I fixed this in the repo.