pulkin/micropython #37: Rebooting the board with Ctrl-D doesn't clear the watchdog

sebi5361, 2020-03-04:
I have noticed that when rebooting the board with `Ctrl-D` the watchdog remains on (if set before).
It might be the same with `machine.reset()`.

If I remember well, this was also the case on some other MicroPython ports but got fixed.

pulkin, 2020-03-04:
A more general question is: how "soft" should soft reset be? Should it

[ ] reset cellular connectivity (disconnect from GPRS, etc)
[ ] reset power setup (remove watchdog, disable GPIOs, disable gps, reset minimal frequency)
[ ] reset time

maybe more?

sebi5361, 2020-03-04:
I believe "soft" should be the same as when powering the board on.
Your list seems exhaustive. I don't think we are missing anything.

I am not that sure, but it seems the reset button is not hard wired to resetting the board.
I say this because it needs to be pressed and release, and in the debugger some kind of error show up at that time. Maybe there is a way to trigger the very same process with the soft reset?

pulkin, 2020-03-05:
Ok, I think it is a good first bug to fix for you or anyone else willing to contribute. What you need to do is:

1. Ensure it is convention (i.e. hardware watchdog is shut down on software reset for mp running on ESP8266/32/any other mainstream board);
2. Go to file `modmachine.c` and put `modmachine_watchdog_off` into `modmachine_init0`;
3. Put the declaration of `modmachine_watchdog_off` before `modmachine_init0`;
4. Build it and test if it works on the board;
5. PR here.

sebi5361, 2020-03-07:
😎

closed 2020-03-07 by sebi5361