pulkin/micropython #38: Example on how to automatically connect to the APN

sebi5361, 2020-03-04:
Could you provide an example on how to automatically connect to the APN when booting?
I implemented a loop but often it just won't connect forever.

pulkin, 2020-03-14:
You may try something like:
```python
import cellular
import time
timeout = time.ticks_ms() + 1e4
while time.ticks_ms() < timeout:
    try:
        cellular.gprs("apn", "user", "pass")
        break;
    except CellularError:
        if time.ticks_ms() < timeout:
            time.sleep(1)
        else:
            raise
```

pulkin, 2020-05-04:
I added a callback `on_status_event(f)` for your convenience. The function you input will be triggered each time the network status changed. You may invent something like
```python
def f(status):
    if status == 1:
        cellular.gprs("apn", "user", "pass")
    cellular.on_status_event(None)
cellular.on_status_event(f)
```
and it will *hopefully* work.

pulkin, 2020-05-09:
MWE https://github.com/pulkin/micropython/blob/master/ports/gprs_a9/examples/example_42_event_driven_gprs.py

pulkin, 2020-06-21:
Feel free to reopen.

closed 2020-06-21 by pulkin