pulkin/micropython #39: Providing a way to tell if gps.get_location() is the actual position

sebi5361, 2020-03-04:
Currently `gps.get_location()` will return the latest location if it cannot get the current location.
There should be a way to tell if `gps.get_location()` is the actual position.

> Both time and location are from NMEA RMC sentence which probably means that they correspond to each other.
> I think about giving access to the timestamp of the last reported location (if there is any). Unfortunately, GPS does not really provide something on demand: it just spams UART2 with its messages

sebi5361, 2020-03-05:
I have noticed that at least one satellite needs to be tracked `gps.get_satellites()[0] > 0` for `gps.get_location()` to give the actual location.

Regarding time, it is not that clear to me how many satellites need to be visible `gps.get_satellites()[1] > 0` to get the actual time. But for sure `gps.time() < 2525040000` to be valid.

bokolob, 2020-08-26:
Now it can be checked with nmea_data() method. Just check if gps has fix and fix quality.  I reckon this issue should be closed