pulkin/micropython #40: BUG: gps.time() rolls back to zero after mqtt.publish(topic, message)

sebi5361, 2020-03-04:
I have noticed that `gps.time()` rolls back to its original value `2525040000` after executing `mqtt.publish(topic, message)`.
I believe the GPS resets completely in that situation.

pulkin, 2020-03-04:
Maybe power?

sebi5361, 2020-03-04:
I am not that sure it comes from my power source.
Could you try that code?
```python
import cellular
import gps
from umqtt.simple import MQTTClient
from time import sleep

cellular.gprs("tm", "", "") # Adjust with your APN

gps.off()
gps.on()
gps.time() # 2525040000
sleep(4)
gps.time() # 2525040004

mqtt = MQTTClient("A9G", "test.mosquitto.org")
mqtt.connect()
gps.time() # >2525040004
sleep(4)
gps.time() # Rolls back to 2525040000
mqtt.disconnect()
```

sebi5361, 2020-03-05:
Did you have time to try this simple code and confirm the bug?

sebi5361, 2020-03-05:
I confirm that my GPS module resets (`gps.get_satellites() == (0, 0) and gps.time() == 2525040000 and gps.get_location() == gps.get_last_location()`) after issuing `mqtt.connect()` or `mqtt.publish(...)`.

...thus making it impossible to send the actual GPS location by MQTT at reasonable rate, as one has to wait for the GPS module to start over each time.

I think this is not related to `mqtt` only: once my GPS got reset after simply importing a module.

sebi5361, 2020-03-11:
Dear Pulkin,
Did you experience the same issue too? Are you planing on solving this bug soon?
Thanks,

pulkin, 2020-03-11:
Yes, I will investigate it but it is not at high priority right now. In the meantime I suggest you to spy the GPS module by reading from its UART2 which should be possible at this point. The only problem I can think of right now is a power issue: mqtt is a pure-python module which is not even aware of this port capable doing GPS.

sebi5361, 2020-03-11:
Indeed I understand there is no reason why MQTT would be responsible for the GPS to reset.
I will spy UART2 to see if I can find additional information, and test with a lab bench power supply to see if the problem persist.
In the meantime if you could just try on your board if you encounter such a bug, it would help me to decide if I need to buy an additional board~~

sebi5361, 2020-03-12:
With a lab bench power supply my GPS still resets...
I really wonder if it is for my board only or for all the boards??
Do you know the baud rate and the settings of UART2 to spy it?

pulkin, 2020-03-12:
It should be the default 115200 (or maybe 9600). You first open the port and then turn on the gps.

sebi5361, 2020-03-12:
The correct baud rate is 9600.

I have tried using `VUSB` with 5V and `VBAT` with 4V, with I>2A but it didn't change anything: I am getting gibberish on UART2 TX regularly. Nothing special is emitted on the RX line when the gibberish occurs as you expected.

I think gibberish occur with every GPRS emissions. Maybe it has to do with interferences. I hope not all the boards are like mine...

Here is an example of what I spy (I have added blanks lines and a return line after the gibberish for better lisibility):
```
...
$GNGGA,000009.029,4337.2350,N,00350.5428,E,0,0,,103.1,M,49.7,M,,*5C
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$BDGSA,A,1,,,,,,,,,,,,,,,*0F
$GPGSV,1,1,00*79
$BDGSV,1,1,00*68
$GNRMC,000009.029,V,4337.2350,N,00350.5428,E,0.000,0.00,060180,,,N*5F
$GNVTG,0.00,T,,M,0.000,N,0.000,K,N*2C

$GNGGA,000010.029,4337.2350,N,00350.5428,E,0,0,,103.1,M,49.7,M,,*54
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$BDGSA,A,1,,,,,,,,,,,,,,,*0F
$GPGSV,1,1,00*79
$BDGSV,1,1,00*68
$GNRMC,00001ðÿþÿÿþÿÿ

$GNGGA,000000.029,4337.2350,N,00350.5428,E,0,0,,103.1,M,49.7,M,,*55
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$BDGSA,A,1,,,,,,,,,,,,,,,*0F
$GPGSV,1,1,00*79
$BDGSV,1,1,00*68
$GNRMC,000000.029,V,4337.2350,N,00350.5428,E,0.000,0.00,060180,,,N*56
$GNVTG,0.00,T,,M,0.000,N,0.000,K,N*2C

$GNGGA,000001.029,4337.2350,N,00350.5428,E,0,0,,103.1,M,49.7,M,,*54
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$BDGSA,A,1,,,,,,,,,,,,,,,*0F
$GPGSV,1,1,00*79
$BDGSV,1,1,00*68
$GNRMC,000001.029,V,4337.2350,N,00350.5428,E,0.000,0.00,060180,,,N*57
$GNVTG,0.00,T,,M,0.000,N,0.000,K,N*2C
...
```
If you confirm this issue doesn't occur on your board I will buy a new one.

sebi5361, 2020-03-12:
Good news for me: by placing the GPS antenna as far as possible from the GPRS antenna, the GPS module doesn't reset anymore. Maybe my GPS antenna has a weak electromagnetic protection. Hopefully other boards won't encounter this issue.

closed 2020-03-12 by sebi5361

pulkin, 2020-03-12:
I thought about this. For me shit mostly happens when I shortcut GSM antenna and one of the nearby pins.