pulkin/micropython #42: "CellularError: Network is not available" when function is called from main.py

BeLikeSohan, 2020-03-14:
Trying to autostart my code when the board starts or resets.
Looks like main.py is executed before the actual board is ready.
I can't find any example or docs for how to make the code autostart.

pulkin, 2020-03-14:
Duplicate of #38. The board is ready but your network is not available yet. It may happen both when starting up but also during normal operation when you leave cellular coverage, for example. Right now I suggest to check `is_network_registered` during startup and to start connecting to GPRS data services only once it returns True.

pulkin, 2020-05-04:
I added a callback `on_status_event(f)`: `f` will be triggered each time network status changes (connect/disconnect, gprs, SIM card inserted, etc). That might be helpful for you.

pulkin, 2020-06-21:
Feel free to reopen.

closed 2020-06-21 by pulkin