pulkin/micropython #43: cellular.call answer not implemented

ahmadmicro, 2020-03-15:
There is no way to answer an incoming call.

pulkin, 2020-03-15:
I think it is easy enough to be implemented by someone else.

1. Go to [micropython docs](https://docs.micropython.org/en/latest/develop/cmodules.html) and familiarize yourself with module layout.
2. Go to modcellular.c and look at, for example, `modcellular_get_imsi`. Something similar has to be implemented here.

https://github.com/pulkin/micropython/blob/892cf5203f1c320afb802c5c5efe30917448a46a/ports/gprs_a9/modcellular.c#L834-L850

3. Copy-paste the function. Remove the body and rename all `modcellular_get_imsi` into `modcellular_answer` (I suggest to name this function `cellular.answer`).
4. In the function body just run `Call_Answer` as described in [API docs](https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/function-api/call.html).
5. Raise a `CellularError` if it returns False. Otherwise just return `None`.
6. Register the function as a part of `cellular` module by adding a new line in the dict in the basement.

https://github.com/pulkin/micropython/blob/892cf5203f1c320afb802c5c5efe30917448a46a/ports/gprs_a9/modcellular.c#L1211-L1255

7. Build, flash and test whether it works on your module.
8. Add function description to port `README.md`.

Total changes are around 10 lines in a single file.

ahmadmicro, 2020-03-15:
Looks like something I should be able to do. Thanks.

drhisham343, 2023-07-09:
Sorry for being a beginner. How can I build flash after making the nessesary changes using microsoft windows. I noticed that the Makefile having some refferance to csdtk42-linux. I tried to use make command on cmd but it didn`t work. can you help?

camp-easy, 2024-04-04:
> Sorry for being a beginner. How can I build flash after making the nessesary changes using microsoft windows. I noticed that the Makefile having some refferance to csdtk42-linux. I tried to use make command on cmd but it didn`t work. can you help?

just try with linux bro