pulkin/micropython #44: Thonny IDE compatibility

iemmeti, 2020-03-19:
Hi,
thanks a lot for your work!
I tried Thonny IDE with your firmware.
REPL works flawlessly, but file upload/download is impossible, returning a lot of errors/exceptions.
If you are interested about fixing this behaviour, I'll be happy to test, if needed (I m afraid I couldn't do much more than that, because I'm a real newbie in python).
Thanks in advance.

pulkin, 2020-03-19:
That's #34. You could at least dump your exceptions here.

pulkin, 2020-03-19:
Ok, I tried this script:
```python
import cellular
print(cellular.get_imei())
```
And it works perfectly. I strongly suggest to try opening raw REPL first and checking whether you actually connected to UART1 and can issue python statements directly.

iemmeti, 2020-03-20:
Hi,
I discovered my fault: a bad usb-serial module I used until yesterday.
This morning I tried another module (mainly because it was already on my desk) and everything went ok.
Thonny integrated REPL (shell) works with both modules, while file upload/download only with the new one.
Probably the issue is related with bad serial timings/levels or noisy signals.
Another change was pudding powering: with the previous module, power was given via a separate USB supply; with the new one, power is from the usb-serial module.
Thank you again!

pulkin, 2020-06-21:
Feel free to reopen.

closed 2020-06-21 by pulkin