pulkin/micropython #48: WebREPL over GPRS

fadiaburaid, 2020-03-25:
Is it possible to implement something like WebREPL on the A9g but over the GPRS instead of WIFI in the ESP8266.  It will allow for sending remote updates to devices over the internet, it will be of great use in many projects. From what I have read a daemon running in the background which forwards the serial REPL through socket must be set. But initially we need to get the GPRS connected of course.

pulkin, 2020-03-25:
Yes, one may in theory just copy `webrepl.py` and run it with minor modifications. In practice GSM service providers neither provide static IP addresses nor allow incoming connections so a9g will have to initiate a connection by itself.

fadiaburaid, 2020-03-26:
I am aware of the IP address problem however since I am using the MQTT to send and track location I can send the current IP address along after getting it using "get_local_ip()" . I was not aware that incoming connections are not allowed. 
Thanks for your help . I appreciate your effort in porting C libraries to allow more functionalities in Micropython.

pulkin, 2020-03-26:
You may use MQTT for back-connection as well: subscribe to some channel from A9G which you will use for publishing your internet gateway IP and port. A9G, once receives the message from MQTT, attempts to connect to the server specified and to give access to its REPL. It is not secure at all, though.

fadiaburaid, 2020-03-26:
I think it will be easier to implement REPL over MQTT then. I can assign MQTT channel just for REPL. A9G will subscribe to the channel to receive REPL commands. I can set my own commands to start and stop REPL through the MQTT channel. The security depends on how secure is my MQTT.

kindmartin, 2020-08-20:
some other related issue is that many carriers also ad CGN (carrier grade Nat) boxes in line so is not only about dinamic/DHCP address but also a nat translation in the middle so  "get_local_ip()" will give the private ip address one, no the public one you need to initiate from repl client. so again one posible solution is to try to init from the A9G side to a fix ip address, uggly!