pulkin/micropython #51: Explore USB connection

pulkin, 2020-03-29:
There are no `csdk` functions available, however, the module seems to support it while `elf`/`lod` contains `hal_Usb*` functions (signatures are [here](https://github.com/cherryding1/RDA8955_W17.44_IDH/blob/b2b6287e8fa9a9f49b679a357b49beb1c31d821c/soft/platform/chip/hal/src/hal_usb.c)). It would be nice to implement the serial port over USB as a start.

ens4dz, 2020-09-20:
I don't know if this help somehow:
I disconnected the usb cable while burning !
after that it appear in device manager as usb device:

[image]

[image]

ubaldus, 2020-09-21:
I could reproduce the same indeed:

[Sun Sep 20 19:51:46 2020] usb 1-4: new full-speed USB device number 19 using xhci_hcd
[Sun Sep 20 19:51:46 2020] usb 1-4: New USB device found, idVendor=1e04, idProduct=0900
[Sun Sep 20 19:51:46 2020] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Sun Sep 20 19:51:46 2020] usb 1-4: Product: Coolsand
[Sun Sep 20 19:51:46 2020] usb 1-4: Manufacturer: Coolsand Technologies
[Sun Sep 20 19:51:46 2020] usb 1-4: SerialNumber: Host USB

ens4dz, 2020-09-23:
the official tool be able to flash rom from usb
i changed this line in download.ini located in cfg folder
```
[download]
8955Dnload=1
useCom=1
useUsb=1
enableUSB=1
```
[image]

**Warning !!!
it seems erase NV data** 

```
>>> cellular.get_imei()
'000000000000000'

```
it needs battery to be connected, without battery it return I/O error at 3%.

ens4dz, 2020-09-23:
i don't know how to get something from those lines:

https://github.com/cherryding1/RDA8955_W17.44_IDH/blob/b2b6287e8fa9a9f49b679a357b49beb1c31d821c/soft/platform/chip/boot/8809/src/boot_monitor.c#L232

ubaldus, 2020-09-24:
well they seem to be quite interesting, KEYIN_1 is responsable to enable BOOT_MODE_FORCE_MONITOR, indeed if I bridge IO26 (GPIO_26=KEYIN_1) with RST and I plug the usb cable to the computer at least linux is detecting the new usb device! :)

ens4dz, 2020-09-24:
Great news, so is possible now to burn python firmware to the board without USB to TTL  !