pulkin/micropython #52: mmc sd card ?

ens4dz, 2020-04-12:
First, thanks a lot for this port.
i have a newbie question: Is possible to log gps data to sdcard ?

pulkin, 2020-04-12:
I did not try it but they claim it is possible. It should be mounted automatically. Afterwards you just ... log the data into some file or whatever.

ens4dz, 2020-04-12:
after search finnaly I found sdcard mounted  under /t 
it read 16 GB FAT32 memory normaly
### M:\2020\A9G\openCellDb>ampy --port com10 ls /t
/t/log
/t/log_34CE00B3BED8_s.txt
/t/media
/t/record
/t/time_lapse
source:
https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/function-api/file-system.html

arlucio, 2020-05-01:
This can definitely be closed, it's just always mounted on /t, and you can manipulate files on it as you would normally do on other directories.

closed 2020-05-01 by ens4dz