pulkin/micropython #53: Sync time to RTC

arlucio, 2020-05-01:
So, I don't really understand what's going on. I don't see anywhere on the code a way to sync the internal RTC to an external time value. This normally is done in mycropython via the ntptime module in this line:

https://github.com/micropython/micropython/blob/4371c971e3dfb743388ccb493c137a25aa9cdd35/ports/esp8266/modules/ntptime.py#L40

We don't have a machine.RTC() function here like the ESP modules but is definitely possible to do one via https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/function-api/timertc.html

What I don't understand is that sometimes it just automatically sync when i do some ntptime request or connect to the internet, don't really now when, and utime.time() start giving me the right value. Sometimes it jus't dont sync and I have to restart the module and keep trying. 

If I restart the module, connect and try again and so, it will eventually sync and start giving the same right values of ntptime and will stay like that for a while even if the module is restarted.

Is that a way to do tell the module to sync already or it is necessary to change the modutime or modmachine to have access to TIME_SetRtcTime. Btw, I'm using A9 module, not A9G, so I don't have access to gps.time().

pulkin, 2020-05-01:
Yes, I think it is possible. Currently, it is this:

https://github.com/pulkin/micropython/blob/b8485f74066ac4c546b5084c076c25f4bf0b8762/ports/gprs_a9/main.c#L400-L404

and I have no idea what's inside.

pulkin, 2020-05-02:
I added `machine.RTC`. It does not set day of the week and milliseconds.

arlucio, 2020-05-03:
Really thanks @pulkin for the fast response, just tested here and it's working great, no changes needed to use the ntptime module directly. 

There is a difference that is good to notice thought. While when the module connects to the internet will automatically sync RTC with localtime (mostly of the time it works, sometimes it did not synced for me), ntptime.settime() will sync to GMT time. 

Thanks and I think that's all, you can close this issue.

pulkin, 2020-05-04:
I am a bit confused about this: does it automatically sync to ntp or just GSM time?

arlucio, 2020-05-04:
When it connects it automatically sync to the localtime you are in, so mine in Brazil is GMT-3, don't really know where it gets it from as we can't see what's inside but probably it gets it from the apn/cellular carrier from the connection response,, probably there is a timestamp on the response or something like that and it just sync.

Not sure why it does not work for me sometimes,  and even connected I get wront time, but as I also always need GMT+0 time on my project, I just always use ntptime.settime() after connecting to be always on GMT+0 time

pulkin, 2020-05-04:
I see. Right, the API supports time zones while micropython does not.

closed 2020-05-13 by arlucio