pulkin/micropython #54: Issue while trying to update Micropython to last version

arlucio, 2020-05-14:
I'm trying to update Micropython of this port to the last version. 

I encountered this issue [#6065](https://github.com/micropython/micropython/issues/6035) because of the way that errno is defined on the SDK. @pulkin any thoughts about how to solve this?

pulkin, 2020-05-14:
If we believe @pfalcon then you should try commenting that fake `Socket_GetLastError` `errno` macro definition.

arlucio, 2020-05-14:
```
micropython/lib/csdtk42-linux/bin/mips-elf-ld: build/gchelper.o: compiled for a big endian system and target is little endian
micropython/lib/csdtk42-linux/bin/mips-elf-ld: build/gchelper.o: endianness incompatible with that of the selected emulation
micropython/lib/csdtk42-linux/bin/mips-elf-ld: failed to merge target specific data of file build/gchelper.o
```
Any ideas how to solve this? Tried to put -EL on opther makefile flags, to force to compile for little endian but didn't had any success

pulkin, 2020-05-14:
They start setting `AS` and/or `ASFLAGS` somewhere in the history replacing this

https://github.com/pulkin/micropython/blob/4679075199b55fa255ac7a562cc6903d60615754/ports/gprs_a9/Makefile#L118

I would check how `gchelper.s` was compiled with `make -n` and whether the above command line arguments are present. I am also pretty convinced it should be this compiler

https://github.com/pulkin/csdtk42-linux/blob/master/bin/mips-elf-as

arlucio, 2020-05-15:
Thanks @pulkin, so I actually made what you did in this commit 973b9def02284f0013082dff2ba07ee598d06f8d and was able to update to the last Micropython version successfully. Included also uasyncio and seems like it's working properly. 

I'm documenting changes and will try to make a push request soon.

closed 2020-05-15 by arlucio

pulkin, 2020-05-16:
Awesome!