pulkin/micropython #56: Вопрос об uart 0

parafagus, 2020-06-01:
Здравствуйте.
Подскажите есть ли возможность полноценно работать с uart 0?
При попытке получить данные с другого устройства я получаю очень странные результаты. Возможно это связана с тем, что там постоянно работает REPL и пытается обрабатывать ввод? Есть ли возможность отключить REPL?

pulkin, 2020-06-01:
Yep, UART0 is REPL, UART1 is attached to GPS.

1. You may invoke [`uos.dupterm(None, 1)`](https://docs.micropython.org/en/latest/library/uos.html#uos.dupterm) and use UART0 to communicate with whatever. Please do not put it into `boot.py` or `main.py` as you will loose access to REPL and there is no such thing as ['safe mode'](https://docs.micropython.org/en/latest/wipy/tutorial/reset.html#safe-boot) here.
2. Or just use UART1 if no GPS involved in your project.

parafagus, 2020-06-01:
Thank you very much.

closed 2020-06-01 by parafagus