pulkin/micropython #60: Firmware burner

toglader, 2020-06-25:
Since coolwatcher is hard to get working in major Linux distributions (because no support) and whole SDK for A9G seems to be disappeared from internet, it would be great if a firmware burner would be implemented for initial installation of micropython for A9G.

pulkin, 2020-06-25:
Yeah, that'd be awesome and I was also thinking about having something similar to `esptool.py`. The first step is to dump and reverse-engineer the communication protocol.

edvm, 2020-08-16:
@pulkin @toglader would be an option to ask Expressif for documentation about this?

pulkin, 2020-08-16:
Sure, go ahead.

bokolob, 2020-09-04:
Is there a connection between espressive and A9G? I thought a9g is produced by unisoc  (RDA chips) and assembled by AiThinker. Am I wrong?

edvm, 2020-09-04:
@bokolob Whops! (I don't know) would be great to confirm that.

Expressif answered the email I sent them asking for documentation about how to write a firmware into the A9G and the answer was:

> 1.Esptool is used to burn firmware into ESP8266, please refer to the below link:
> https://github.com/espressif/esptool
>
> 2.You mentioned that you want to implement a tool similar to Esptool, I hope the serial protocol can be useful to you.
> https://github.com/espressif/esptool/wiki/Serial-Protocol

It was strange for me that asking for "how to write firmware into the A9G" the answer was "To  burn a firmware into ESP8266", so sounds like you are right @bokolob :sweat_smile:

ens4dz, 2020-09-04:
I think this tool is an other choice to reverse-engineer the communication protocol :
http://wiki.ai-thinker.com/gprs_download

bokolob, 2020-09-04:
I asked AiThinker about support and they said that SDK isn't supported for the moment and they have no plans for support or opensource it 8( Only AT firmware has some kind of support as I get.
As well they don't provide any documentation.

bokolob, 2020-09-04:
Probably they wanted to create a competitor for esp.

stefankranz, 2021-05-19:
Hi, you can use the aithinker firmwareupdate tool 2.1 this works easy.