pulkin/micropython #61: axTLS wrap_socket() blocks execution after some loops

arlucio, 2020-06-29:
I did some tests sending data to a https cloud database using wrap_socket() and it always hangs and completely block execution after some loops. Sometimes after 20, 100, or 200 but it sometime always hangs on wrap_socket() and stays like this until the watchdog resets. As wrap_socket() on axTLS is blocking, there is no much choice on what to do in this case.

Would be great if we could port mbedtls or wolfssl to this port. Mbedtls is already used on esp32 and stm32 ports, but I believe esp32 uses an specific IDF implementation, and just stm32 uses a more generic one.

Wolfssl was not ported yet to micropython but would probably be the best, as it has a dtls implementation (not available in any micropython port yet) and seems like they are working to provide HTTP3/QUIC support, which seems perfect for IoT.

I will probably be working on this sometime in the near future, but any help would be great.

pulkin, 2020-06-29:
I see. I am a little bit more cautious about putting another SSL lib here: I would not spend time on this in the current stage. There is no guarantee that if you face some random crash such as #36 with another lib you will *ever* be able to debug it. Because there are no debugging tools.

Given that, I would appreciate someone to work on more basic things such as making `gdb` working here. It seems to be working with examples but not with `mpy` firmware: and I do not know why.

Kmiz07, 2021-03-05:
Podrian citar un ejemplo que les este funcionando de conexion ssl?