pulkin/micropython #62: How to get coolwatcher working?

bokolob, 2020-07-02:
I use Debian stable (buster) and got Linux coolwatcher from the docker image (I hadn't found any other way to get it). But it doesn't work.
I copied 8955 stuff from windows version to linux, because the last one doesn't have it. 

Last error is
`
`require': no such file to load -- rbconfig (LoadError)
--

`

I've also tried to use wine and windows version of coolwatcher, but it gave  me other error - connection timeout (it seemed that it couldn't connect to coolhost daemon)

Could you give me some piece of advice, and tell me please where I can get a working coolwatcher?

Thanks!

pulkin, 2020-07-02:
https://github.com/pulkin/csdtk42-linux

It is also a submodule in this repo so you might have it already. Unfortunately, I cannot help you with the rest.

bokolob, 2020-07-08:
Thanks! It works :)

closed 2020-07-08 by bokolob