pulkin/micropython #64: Hangs up without uart connection

bokolob, 2020-07-08:
Hello again.
My board hangs up after USB cable disconnecting. It is powered with a lipo battery and connected to computer through a cp2120 USB-UART converter (converter shares ground with the board, but powered by USB).  I upload a simple GPS tracking script with ampy, and run "import tacker" in the minicom. 
It starts to print some stuff out, and then I disconnect the cable. After about 5 minutes (yes, it happens not immediately but after a while) module hangs up and only reset can help. However, the script will work for hours with UART connected.

It's reproducible in 100% of tries.

It's like

`
while True:
     gps.on()
     cellular.gprs(...)
     write_coords_to_socket()
     utime.sleep(60);
`

pulkin, 2020-07-08:
It is not a firmware problem given the way you describe it. My best guess is that you actually connect 5V of your cp2120 to the board. Once disconnected it dies either because your battery does not provide enough power or you use some sort of power bank which goes to sleep if not enough power is drawn.

bokolob, 2020-07-08:
No, it's definitely not the reason...  
I've made an experiment just now - without running program it continues work. Their  +5vs aren't connected and I have 1000 mAh LiPo, I want to believe that it has good quality  :)

And eventually, it has got stuck just when I was writing this answer and being connected to USB! 

Maybe the reason is some network problems? Maybe I should try without GPRS. 

My  socket code is here

```
def send_rtt_coordinates():
    s = socket.socket()
    try:
        s.connect((SETTINGS['rtt_server'], SETTINGS['rtt_port']))
        print(get_rtt_string())
        s.write(get_rtt_string())
    finally:
        s.close()
```

But I don't check if I'm connected to the network or not.

bokolob, 2020-07-08:
The full code is here - https://github.com/bokolob/tracker/blob/master/tracker.py

bokolob, 2020-07-09:
So,  without sockets, it seems to work without any problems.

pulkin, 2020-07-09:
> And eventually, it has got stuck just when I was writing this answer and being connected to USB!

In this case you should probably investigate what `coolwather` reports. You might also have forgotten `else` before

```python
set_gprs_state(False)
```

bokolob, 2020-07-09:
Reports? Do you mean Tracer plugin or gdb? Or it is something else?
How to make tracer work on Linux? It pretends to be connected, but display nothing :(
And about gdb - where can I get elf file that is required?

bokolob, 2020-07-09:
I've added some debug prints and it seems that problem is with sockets. And the strangest thing is that the board become unable to take calls (you get a message that the phone is out of service).

bokolob, 2020-07-10:
My investigation shows that if we don't get a response to connect call, then the board freezes. It's probably because of the lwip call internals.

I tried the next piece of code, but it still doesn't work.

```
def connect_with_timeout(s, server, port, timeout):
    s.setblocking(0);

    try:
        s.connect((server, port))
    except Exception as err:
        pass

    readable,writable,exceptionavailable = select.select([s],[s],[s],10)

    for s in writable:
        s.setblocking(1)
        return s

    raise Exception("Connect timeout")
```

pulkin, 2020-07-10:
Ok, this I can investigate.

bokolob, 2020-07-10:
But you know.. I think it's somehow connected with my battery scheme. 
[image]

bokolob, 2020-07-13:
Finally, I've investigated this issue.  Establishing TCP connection via GPRS producing current spikes, that makes those effects. The root of the problem was the lack of decoupling capacitance.

closed 2020-07-13 by bokolob