pulkin/micropython #66: Make it possible to pass an timeout argument to sms send

bokolob, 2020-08-04:
This   line
WAIT_UNTIL(sms_send_flag, TIMEOUT_SMS_SEND, 100, mp_warning(NULL, "Failed to send SMS. The module will continue attempting sending it"));

blocks code for 10 seconds, but it's not the desired  behavior sometimes

pulkin, 2020-08-05:
Yep, its #15. There is even a guide there.

closed 2020-08-05 by pulkin

bokolob, 2020-08-05:
Ok , I will try to do it.

By the way, why not to pass events to python level and rewrite Cellular in python? Make a thin wrapper for C api. The only problem -  handling events in repl. But it seems to be more convenient and “debuggable”

pulkin, 2020-08-05:
There is already 'on_sms' handler. Otherwise the code is as thin as it could be. Debugging event-driven or async python code is even more difficult, especially for the port where no testing is performed. Timeout implementation similar to 'cellular.GPRS' will cover most use cases.