pulkin/micropython #70: some power consumtions testing and some q if any can address

kindmartin, 2020-08-20:
hi all, pulkin thanks you work in this repo/

here some initial current measurements for a AG9 pudding Ive got last week.

1.- with just usb power 5.1v: (no lipo connected /or connected but full at charge >4.15v)
     ~ 40mA avg / 80mA peak just the module turned on, running normal FREQ/valid SIM inserted)

2.- adding gps.on()
     ~ 80mA avg / 90mA peak just the module turned on, running normal FREQ/valid SIM inserted)

3.- adding cellular no flight mode, low rssi about -87db
     ~ 90mA avg / 130mA peak just the module turned on, running normal FREQ/valid SIM inserted)

4.- adding gprs, low rssi about -87db
    ~ 90mA avg / 130mA peak just the module turned on, running normal FREQ/valid SIM inserted)

5.- data active for agps request, low rssi about -87db
    ~ 200mA avg / 435mA peak during about ~4 seconds.

6.- all off (gps.off. flight mode(1))
   ~ 40mA avg , except a short cell unregister task

7.- with cpu frq minimal / machine.set_min_freq(32768)
    ~ 25 mA avg ,!

now some Q's !

A. what machine.set_idle does ? the int argument is in miliseconds?
B. there is any way to down those 25mA ?
C. if i use a main ESP32 with deepsleep connected to A9G via serial port, there is any way /command to shut down all the module until a configured pin goes up or down? any RTC option for this? perhaps just drive the A9g module Vin power in from a pnp transitor to handle those 400mA maximun?

D. I was using a ESP32 uPy version from Boris Lovosevic, and noted there is possible to include py modules in the fw, see at https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/tree/master/MicroPython_BUILD/components/micropython/esp32/modules
so doing so here could help to bring agps or other modules to the system? how much space do we have not counting with the attached SD ?

E. any idea how to add that missing smd accelerometer LIS3DHx? I thnk is wired to ic2 port so a python driver could just work. I'm trying to get that chip and solder in my pudding. but dont know if it will works.

pulkin, 2020-08-20:
`set_idle` tweaks CPU frequency. It should be possible to reduce current draw by switching flight mode on or even disconnecting the SIM card. I did not explore this. There is an `modmachine.off` function but it never worked for me. I did not explore hardware options but others certainly used atmega to switch A9 on and off. It is possible to freeze modules in the fw. Please google for hw specs and board layout.

kindmartin, 2020-08-21:
thanks! I'm trying to do something useful in this quarantine time. so tried too modmachine.off no success neither.
a couple more hints if you can help here.

I think all this not related to initial post from I, perhaps better create a new one.

1.- what could be a timer lib in case no hw timers are available in the A9G platform. tryed BlynkTimer but is up to seconds timeslots so not good to let a blinking led inform statuses.

2.- How could I freeze for instance urequests module to avoid download?
# Import requests (download if necessary)
try:
    import urequests
except ImportError:
    import upip
    upip.install("micropython-urequests")
    import urequests 

3./ where do I have to drop valid boot.py and main.py ? I tried at /t SD root but do no auto-execute in the boot/reset.

thanks in advance.

bokolob, 2020-08-24:
Drop main.py to root (/)

bokolob, 2020-08-24:
There is micropython.schedule lib, but I’ve never tried it. A9g supports timers, but they are not ported yet

bokolob, 2020-08-24:
By the way, how did you make agps request? As I remember it’s not ported to mp. Don’t you mean lbs?

kindmartin, 2020-08-24:
thanks for reply, there is an example of agps in / examples. https://github.com/pulkin/micropython/blob/master/ports/gprs_a9/examples/example_21_agps.py
instead of use upip at running code Ive got the lib by other means and put that one at /t/lib for import.

os.chdir("/t/lib")
import agps
here the lib im using works perfect. originally i was some urequests lib typo, fixed in this attached.

[agps.zip](https://github.com/pulkin/micropython/files/5120216/agps.zip)

bokolob, 2020-08-24:
Mm, here is some problem with naming ;) agps is downloading dictionary with satellite’s fresh coordinates. While lbs is getting location by cellular network. Agps is supported  by a9g firmware, code (with very bad smell btw) can be found on their github

bokolob, 2020-08-24:
You can find schematic with accelerometer on a9g doc page

kindmartin, 2020-08-24:
I see, https://stackoverflow.com/questions/11760962/what-is-gps-agps-lbs-in-android

in my case, as an starter having a possible fast gps fix could be a plus, but least to have an idea based on cellular.stations() info is good too.

kindmartin, 2020-08-24:
> You can find schematic with accelerometer on a9g doc page

yes I founded also some post https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/102 so now I will try to buy a couple to add in my boards. not easy to get where Im from in this covit 19 times.

M

bokolob, 2020-08-25:
You can buy it on Alibaba, board with accel as well as any spi accel. You can try i2c but I couldn’t make it works.

kindmartin, 2020-08-25:
good comment to save buy /wait/ pray as in my country Argentina those aliexpress /china imports dont come easy.

 I had this attached lib to test . [lis3dh.zip](https://github.com/pulkin/micropython/files/5126437/lis3dh.zip)

but thinking batter to go for ttgo Tcall as those are ESP32 easy to go deepsleep and add an gps/mpu just work with loboris upy easu to compile/customize firmwares. I woul like to help here with pending portings like machine.Timer, PWM, sleep modes, etc but im too new in c/python lib programing.

[image]

bokolob, 2020-08-26:
Do you know that the official sdk is dead? I mean nobody supports it and it’s future is very foggy. If you are still interested -  take any issue :)