pulkin/micropython #77: Using the SD card socket available in PUDDING board?

wzab, 2020-09-06:
Hi,
The pudding board is equipped with the SD socket. It would be nice to store it the additional Python code.
It would reduce the wear of internal FLASH memory and speed-up installing and editing the bigger Python scripts (that could be edited in the SD card itself).
Is it possible to access it via the hardware SD controller or via hardware SPI?
What should I add or enable in the ports/gprs_a9/mpconfigport.h to achieve that?
Has anybody tried that?

TIA & Regards,
Wojtek

pulkin, 2020-09-06:
That card is mounted under `/t`. I do not know what is specifically supported, you may try looking [here](https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/function-api/file-system.html).

bokolob, 2020-10-03:
I've been using a sdcard for logging for a couple of months. Everything is ok, the card is mounted under /t, and it has to be in fat32.