pulkin/micropython #78: Using the PIN-protected SIM card in A9G?

wzab, 2020-09-06:
In case if A9G is used in an unattended remote sensing device, there is a legal risk of abuse of SIM card extracted from the stolen device.
That risk may be slightly mitigated by use of PIN-protected SIM card (of course it is still possible to sniff the PIN sent to the SIM card, but it is much more complicated than simply taking the SIM card and placing it in another device).
I can see that A9G supports the PIN-related AT commands (AT+CPIN and others), but I can't see any way to access them from micropython. Is it possible to add such functionality? Or maybe I have overlooked something?

Best regards,
Wojtek

pulkin, 2020-09-06:
No there is nothing related to PIN here.

wzab, 2020-09-06:
Can it be added? Where should I look for any possibility to add it?
Regards,
Wojtek

pulkin, 2020-09-06:
Yes it can be added but it also requires a lot of work beyond mpy. [This](https://github.com/cherryding1/RDA8955_W17.44_IDH/blob/b2b6287e8fa9a9f49b679a357b49beb1c31d821c/soft/at/ats/sim/src/at_cmd_sim.c#L850) is how the modem command is implemented.