pulkin/micropython #79: not a issue,is possible to fix IMEI ?

ens4dz, 2020-09-06:
I have 3 of GSM A9 Pudding Board, 2 of them lost their IMEI ! :
but still, GSM calls and SMS are working fine !
i flashed the smaller size ROM **firmware_debug_fla­sh.lod** with : **Ai-Thinker firmwarw_tool_v2.1**­ But they failed to boot. Then, i flashed the **firmware_debug_ful­l.lod**

the first one:
```
AT+GSN
000000000000000
```

the second one:
```
>>> cellular.get_imei()
'000000000000000'
```

the third one is fine:
```
>>> cellular.get_imei()
'8687150300xxx00'
```

pulkin, 2020-09-06:
I do not know. This is what the assembly shows:
```c
INT32 tgt_GetImeiSv(UINT8 simId,UINT8 *imeiSv) {
...  
  if (((imeiSv != (UINT8 *)0x0) && (simId < 4)) &&
     (IVar1 = FACTORY_ReadField((uint)simId * 9 + 4,9,imeiSv), IVar1 != 0)) {
    hal_DbgAssert(&UNK_881ca800);
  }
...
  return IVar1;
}
```
```c
INT32 FACTORY_ReadField(UINT32 offset,UINT32 size,UINT8 *buff)

{
...
  if (g_factory_info.has_init != '\0') {
    factory_SemWaitFor();
    SUL_MemCopy8(buff,offset + 0x82094f04);
    factory_SemRelease();
  }
  ...
}
```
So maybe if `0x82094f04 + 0x4` is a part of SPI memory or any other memory that you can write you will be able to change your IMEI.

Even if non-zero, IMEI of these boards are just random and fail the checksum anyway.

ens4dz, 2020-09-06:
So, it is ok to leave off that zeros IMEI .
Thank you for this awesome port project and for your great support. I wish you the best.

closed 2020-09-14 by ens4dz

ens4dz, 2021-08-11:
Update:
When i read an old tutorial, I found that
Is possible to write IMEI by AT firmware:
AT+EGMR=**1**,7,................
To Read IMEI
AT+EGMR=**2**,7
 IMEI is stored at the end in file named:
 AT_CFG_0.BIN
Also, IMEI is printed as QR code at module shield, you need good phone qr to recognize it
I che
[image]