pulkin/micropython #81: /t/main.py

ubaldus, 2020-09-09:
Would it make sense to add a /t/main.py check and eventual execution on modules/_boot.py by default? 

I mean I think that is the spirit of boot.py and main.py so having in this case an sd card would be a pity not to implement it :)

pulkin, 2020-09-09:
What kind of problem this solves?

ubaldus, 2020-09-09:
The idea is that I could rewrite the main.py on the sd card without having to burn again the whole firmware nor risking to mess up with the internal storage

pulkin, 2020-09-09:
Do you say you have problems rewriting `/main.py` as a part of internal SPI memory?

ubaldus, 2020-09-09:
No, my idea is that I burn the firmware, so the device is ready and working, I can then pack it into a nice case but anytime I want to change something I will just need to rewrite the /t/main.py on the sd card without having to take the board out and burn a new firmware.

ubaldus, 2020-09-09:
or having to connect via serial which is also not as easy as to connect via wifi for instance on the esp32...

pulkin, 2020-09-09:
You may put `/main.py` with
```python
sys.path.append("/t")
import main_from_sd_card
```
before packing into a nice case. Then, change to whatever you want in `/t/main_from_sd_card.py`.

bokolob, 2020-09-09:
By the way you don't have to re-burn firmware to update main.py, you can use ampy,  and upload files through UART.

ubaldus, 2020-09-09:
of course Pulkin I could do it so but I would then need to connect the board to the HST pins, burn the firmware, then connect the uart, write the /main.py and disconnect which is fine if I am doing on a board but what if I am flashing hundreds?

I mean I could also easily write my own version of modules/_boot.py I was just wondering if a similar feature could be helpful to someone else and as micropython already is anyway looking for a /boot.py and a /main.py why not adding for this specific board support for a /t/main.py? :)

bokolob, 2020-09-09:
Why do you have this strong desire of burning firmware?  It's not necessary.. )

pulkin, 2020-09-09:
> but what if I am flashing hundreds?

Then you may update `_boot.py` with your code, rebuild the firmware and flash it hundred times.

If you provide an example of at least one other mainstream `mpy` firmware with this behaviour I might want to look into it.

ens4dz, 2020-09-22:
Yesterday, I put machine.restart()  line at boot.py !
It's takes me a while to reburn firmware
I think it will be great to port it.

https://docs.micropython.org/en/latest/pyboard/general.html

https://github.com/micropython/micropython/issues/1915

ubaldus, 2020-09-22:
Question is now, should we run /boot.py and /main.py anyway first if they are present?

```
    // Startup scripts
    pyexec_frozen_module("_boot.py");
    pyexec_file_if_exists("boot.py");
    if (mp_vfs_import_stat("/SKIPSD") == MP_IMPORT_STAT_NO_EXIST) {
        pyexec_file_if_exists("/t/boot.py");
    }
    if (pyexec_mode_kind == PYEXEC_MODE_FRIENDLY_REPL) {
        pyexec_file_if_exists("main.py");
        if (mp_vfs_import_stat("/SKIPSD") == MP_IMPORT_STAT_NO_EXIST) {
            pyexec_file_if_exists("/t/main.py");
        }
    }
```

ens4dz, 2020-09-22:
> Question is now, should we run /boot.py and /main.py anyway first if they are present?
>
> ```
>     // Startup scripts
>     pyexec_frozen_module("_boot.py");
>     pyexec_file_if_exists("boot.py");
>     if [...]

take a look here:
https://github.com/micropython/micropython/blob/42342fa3cb30e2eac56ceb1d21b4eb60a0f406f3/ports/stm32/main.c#L624