pulkin/micropython #84: Where is the compiled fw .lods files to burn after cd ports/gprs_a9 + make

kindmartin, 2020-09-11:
I need some help to re flash my A9G module as it do not respond over over UART1 (TX1/TX2), I already did burn weeks ago a micropython firmware on the module. I think yesterday i used the usb 2serial at 5v not 3v3 and perhaps I broke the UART1. ?? the point is that only host tx/rx port works with coolwatcher/or fw burner,

I had both windows CoolWatcher burner / ubuntu toolchain enviroments ready but im not sure where output directory.

I need some help to re flash my A9G module as it do not respond over over UART1 (TX1/TX2) 
 thanks in advance.

M

pulkin, 2020-09-12:
```bash
~ ls hex
firmware_debug_final.elf  firmware_debug_final.map  firmware_debug_flash.lod  firmware_debug_full.lod
```

kindmartin, 2020-09-14:
Thanks Pulkin! my module after a successful Host UART does no respond at UART1. any way to compile a new .lod firmware redirecting repl to lets say IO2/IO3 as I do not need CTS/RT or other at IO20-IO30?
Im from Argentina an have a new pudding take like 2 months to more.

thanks in advance.

M

pulkin, 2020-09-14:
You may try first UART2. Otherwise one needs to implement software UART.

kindmartin, 2020-09-14:
Thanks, but UART2 doesnt  connected to GPS port?
actual micropython port does support SW UARTs in other pins?

pulkin, 2020-09-14:
Nope.