pulkin/micropython #89: less performance ?

ens4dz, 2020-09-18:
Before months I run this benchmark test:
```
import time
def performanceTest():
    millis = time.ticks_ms
    endTime = millis() + 10000
    count = 0
    while millis() < endTime:
        count += 1
    print("Count: ", count)
performanceTest()
```

I got:
```
MicroPython v1.9.4-2783-g8c6c7d7fa-dirty on 2020-06-21; A9/A9G module with RDA8955
performanceTest()
Count:  450712
machine.set_min_freq(machine.PM_SYS_FREQ_312M)
performanceTest()
Count:  446741
machine.set_idle(1)
performanceTest()
Count:  122113
```

**But now:**
--------------
```
MicroPython v1.9.4-2810-gf467d36d0-dirty on 2020-09-12; A9/A9G module with RDA8955
>>> machine.set_min_freq(machine.PM_SYS_FREQ_312M)
>>> performanceTest()
Count:  284251
>>> import cellular
>>> cellular.flight_mode(1)
True
>>> performanceTest()
Count:  307978
>>> machine.set_idle(1)
>>> performanceTest()
Count:  106629
```

even in old copy firmware:
```
MicroPython v1.9.4-1914-gb8485f740-dirty on 2020-04-08; A9/A9G module with RDA8955
performanceTest()
Count:  283614
------------

```

ens4dz, 2020-09-18:
link:
[2020-06-21 firmware](https://drive.google.com/uc?id=1CPM9L9fcGUR6Mz9Eq26sGQ7e2kuYDyzS&export=download)

pulkin, 2020-09-19:
I have no time to investigate this. You may use `git bisect` to identify the commit when the performance drop emerged.

ens4dz, 2020-09-21:
> I have no time to investigate this. You may use `git bisect` to identify the commit when the performance drop emerged.

I have learned a lot from you, thank you for tips.
this what i had after do "git bisect"

```
m7@m7-VirtualBox:~/micropython$ git bisect start 
m7@m7-VirtualBox:~/micropython$ git bisect good 8c6c7d7fa
m7@m7-VirtualBox:~/micropython$ git bisect bad
Bisecting: 13 revisions left to test after this (roughly 4 steps)
[5b2da515da07fa6e45daede08051d3b49d29d24a] cellular: SMS.delete_by_index -> SMS._withdraw_by_index
m7@m7-VirtualBox:~/micropython$ git bisect bad
Bisecting: 6 revisions left to test after this (roughly 3 steps)
[e6652257c80a4bd4f591f6625fe1ae3c08727231] cellular: add timeout to sms_send
m7@m7-VirtualBox:~/micropython$ git bisect good
Bisecting: 3 revisions left to test after this (roughly 2 steps)
[6478ae62342671c830b30b508922355775327b9d] deleting sms by index (useful for cleaning up whole memory)
m7@m7-VirtualBox:~/micropython$ git bisect bad
Bisecting: 0 revisions left to test after this (roughly 1 step)
[c846c69ab31e85934756d3df9d1e136dc6718f9f] py: define ATOMIC macros
m7@m7-VirtualBox:~/micropython$ git bisect bad
Bisecting: 0 revisions left to test after this (roughly 0 steps)
[c710f567bc4d951bf39697017f85f91373311d2c] cellular: add SMS.get_storage_size
m7@m7-VirtualBox:~/micropython$ git bisect good
c846c69ab31e85934756d3df9d1e136dc6718f9f is the first bad commit
commit c846c69ab31e85934756d3df9d1e136dc6718f9f
Author: Artem <gpulkin@gmail.com>
Date:   Sat Aug 29 23:44:57 2020 +0200

    py: define ATOMIC macros

:040000 040000 1bb1573976dcf39c4899fba27c63ad7d6864571d 00d57dc331edc4785776ee3a0bdc1b1c2bc6725b M	ports
```

ens4dz, 2020-09-21:
```
MicroPython v1.9.4-2797-g5b2da515d-dirty on 2020-09-21; A9/A9G module with RDA8955
Type "help()" for more information.
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import time
=== import machine
=== def performanceTest():
===     millis = time.ticks_ms
===     endTime = millis() + 10000
===     count = 0
===     while millis() < endTime:
===         count += 1
===     print("Count: ", count)
=== performanceTest()
Count:  297334
>>> performanceTest()
Count:  298183
>>>
```

```
MicroPython v1.9.4-2790-ge6652257c-dirty on 2020-09-21; A9/A9G module with RDA8955
Type "help()" for more information.
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import time
=== def performanceTest():
===     millis = time.ticks_ms
===     endTime = millis() + 10000
===     count = 0
===     while millis() < endTime:
===         count += 1
===     print("Count: ", count)
=== performanceTest()
===
Count:  445449
>>> performanceTest()
Count:  477098
```

```
MicroPython v1.9.4-2793-g6478ae623-dirty on 2020-09-21; A9/A9G module with RDA8955
Type "help()" for more information.
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import time
=== def performanceTest():
===     millis = time.ticks_ms
===     endTime = millis() + 10000
===     count = 0
===     while millis() < endTime:
===         count += 1
===     print("Count: ", count)
=== performanceTest()
===
Count:  295734
>>> performanceTest()
Count:  297690
```

```
MicroPython v1.9.4-2792-gc846c69ab-dirty on 2020-09-21; A9/A9G module with RDA8955
Type "help()" for more information.
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import time
=== def performanceTest():
===     millis = time.ticks_ms
===     endTime = millis() + 10000
===     count = 0
===     while millis() < endTime:
===         count += 1
===     print("Count: ", count)
=== performanceTest()
===
Count:  296928
>>> performanceTest()
Count:  297241
>>>
```

```
MicroPython v1.9.4-2791-gc710f567b-dirty on 2020-09-21; A9/A9G module with RDA8955
Type "help()" for more information.
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import time
=== def performanceTest():
===     millis = time.ticks_ms
===     endTime = millis() + 10000
===     count = 0
===     while millis() < endTime:
===         count += 1
===     print("Count: ", count)
=== performanceTest()
===
Count:  436714
>>> performanceTest()
Count:  487339

```

ens4dz, 2020-09-21:
I can confirm that commit: https://github.com/pulkin/micropython/commit/c846c69ab31e85934756d3df9d1e136dc6718f9f do performance drop. 
does it worth it ?

```
MicroPython v1.9.4-2811-g04411c939-dirty on 2020-09-22; A9/A9G module with RDA8955
Type "help()" for more information.
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import time
=== def performanceTest():
===     millis = time.ticks_ms
===     endTime = millis() + 10000
===     count = 0
===     while millis() < endTime:
===         count += 1
===     print("Count: ", count)
=== performanceTest()
===
Count:  481845

```

bokolob, 2020-09-30:
Hm, It was my commit :)
I plan to implement threads and interrupt support, and it will be necessary :(

bokolob, 2020-09-30:
I'm curious what inside your benchmark can use MICROPY_BEGIN_ATOMIC_SECTION so heavily?

pulkin, 2020-10-16:
It might for example be memory allocation for `""Count: "`.