pulkin/micropython #91: Pudding operating from Battery shutdown after boot/

kindmartin, 2020-10-15:
Hi all, these days I got some more pudding boards and i wanted to continue from where I was when I broke the original pudding board.

I was able to compile and burn the firmware using coolwatcher so now the pudding work ok from usb power. 
but when I try to boot from a 4.2v lipo battery feeding VBAT then PWR button the board boot but after 2 o3 seconds shutdown. perhaps the compiled firmware is not ok?

The idea is to use a ESP32 / WROOM32 on top of A9G as i need wifi and a better sleepmode / RTC features that are available in those loboris/or general ESP32 firmwares, and use a ESP UART conected to A9G main UART as slave to access GPS data and some 2G cellular capabilities as SMS/calls. below a picture of those ones connected as prototypes.

The idea is after ESP32 boot from a deepsleep, it will power-on the pudding via a MOSFET or just better solution via EN pin or even better via a upy command if available in future. but I need to power both from pudding VBAT.

I tested VBAT on a new other pudding and it happend the same. what Im missing?

---->>>>
Disconnected from remote host(Ai thinker A9G - Copy) at 00:26:01.
Type `help' to learn how to use Xshell prompt.

Connecting to COM6...
Connected.
??oPython v1.9.4-2792-gc846c69ab-dirty on 2020-08-29; A9/A9G module with RDA8955
Type "help()" for more information.
>>> 

(then pudding shutdown silently)

[image]

kindmartin, 2020-10-15:
I found what was happening, I was pressing PWR for too short time. I PWR is presed for more than 3 seconds, it boots ok from VBAT.
hope this help if someone related to the issue or the ESP32+A9G idea.

M

ens4dz, 2020-10-15:
As the picture above, is possible to make calls without A9 core ?!!

kindmartin, 2020-10-15:
yes I think so. the main rational here is that in my usecase wifi is a must, also deepsleep. 
now Im getting ~0.1 mA in deepsleep. I will try to create a lib to manage A9G as slave via UART passing data from ESP32 back and forth, not sure if gprs will work but surely those a9g libs will works.

anyone want to join this aproach it will be welcome!

[image]

pulkin, 2020-10-16:
It is great to find two modules speaking REPL but I would probably use the AT firmware for stability reasons. I would at least try to.

closed 2020-11-30 by kindmartin