pulkin/micropython #92: mfrc522 SPI RFID reader

fadiaburaid, 2020-10-16:
I am trying to use mfrc522 RFID reader with the A9g.

I have found this Micropython class [https://github.com/wendlers/micropython-mfrc522](url)
It only supports WiPy, LoPy ,FiPy and ESP8266.
I know there is support for software SPI on A9g.

I assume I have to do changes on the class to get this to work.
Can you please advice me on changes ?

On  read.py
`
def do_read():

	if uname()[0] == 'WiPy':
		rdr = mfrc522.MFRC522("GP14", "GP16", "GP15", "GP22", "GP17")
	elif uname()[0] == 'esp8266':
		rdr = mfrc522.MFRC522(0, 2, 4, 5, 14)
	else:
		raise RuntimeError("Unsupported platform")

	print("")
	print("Place card before reader to read from address 0x08")
	print("")
`

On mfrc522.py 
`
def __init__(self, sck, mosi, miso, rst, cs):

		self.sck = Pin(sck, Pin.OUT)
		self.mosi = Pin(mosi, Pin.OUT)
		self.miso = Pin(miso)
		self.rst = Pin(rst, Pin.OUT)
		self.cs = Pin(cs, Pin.OUT)

		self.rst.value(0)
		self.cs.value(1)
		
		board = uname()[0]

		if board == 'WiPy' or board == 'LoPy' or board == 'FiPy':
			self.spi = SPI(0)
			self.spi.init(SPI.MASTER, baudrate=1000000, pins=(self.sck, self.mosi, self.miso))
		elif board == 'esp8266':
			self.spi = SPI(baudrate=100000, polarity=0, phase=0, sck=self.sck, mosi=self.mosi, miso=self.miso)
			self.spi.init()
		else:
			raise RuntimeError("Unsupported platform")
`

bokolob, 2020-10-17:
I use spi in my project. You can use it as an example
https://github.com/bokolob/tracker/blob/06031a7c42dfbbece5f73f1c55427009deb97990/lis3dsh.py#L161

fadiaburaid, 2020-10-17:
> I use spi in my project. You can use it as an example
> https://github.com/bokolob/tracker/blob/06031a7c42dfbbece5f73f1c55427009deb97990/lis3dsh.py#L161

Thanks for your help. So I can assign the pins for software SPI even though they are not SPI pins.

bokolob, 2020-10-17:
I have software i2c either 💯