pulkin/micropython #96: SMS Error object has no attribute 'SMS_RECEIVED'

marcosasilvalepe, 2020-11-09:
Hi! I've been trying to read SMS using example_41_sms.py but when the SMS is received I get the following error:

Traceback (most recent call last):
  File "/t/main.py", line 16, in sms_handler
AttributeError: 'module' object has no attribute 'SMS_RECEIVED'

Has anyone had this problem ?

pulkin, 2020-11-09:
That example might be outdated since e7bfbb23e4f86d6d9dd05776cda569ba0d7aa275.

marcosasilvalepe, 2020-11-09:
OK, got it!

Doing `else` insead of `elif` solved it. Thanks.

closed 2020-11-09 by marcosasilvalepe

iotmj, 2023-10-03:
Hi Dracoit, how you fix the above , i also get warning failed to poll all sms, list may incomplete. after this 'none'. i executed                   
 else:    
        print("SMS received, attempting to read ...")
        ls = cellular.SMS.list()
        print(ls[-1])
        flag = 0