pulkin/micropython #99: New (?) csdk

bokolob, 2020-11-20:
I’ve got links from ai-thinker support team. There are toolchain and csdk. Could you check that sdk is fresher?

https://drive.google.com/file/d/1gtX49BgB3DV7nkznwRB_U0dYJr_1P-ZE/view?usp=sharing 

https://drive.google.com/file/d/1Rv2YKTfL7hnydce-15DwEyJqKBgo4qTO/view?usp=sharing

pulkin, 2020-11-21:
Second link is dead.

ens4dz, 2020-11-22:
Same link:
https://drive.google.com/u/0/uc?id=1Rv2YKTfL7hnydce-15DwEyJqKBgo4qTO&export=download

Also, there are a copy in telegram channel

jcherrabi, 2020-11-22:
Any idea on how to compile this new SDK? i'm using windows... and i noticed missing the batch files that sets up the environment... any help is appreciated.. thanks for sharing 👍

pulkin, 2020-11-22:
@ens4dz , `lod` files are different. Both are under the same name `SW_V2131_csdk.lod` though. sha256:

`c6b0b4618308b3e6e4b4c36e551d149b83e6a9185e8ab17a6e3ace743af70560` -- current
`411ac840264be2f546b9589952d368cd9c10bce61c6d3c71c37e00d3c69d41d8` -- yours

bokolob, 2020-12-03:
> Any idea on how to compile this new SDK? i'm using windows... and i noticed missing the batch files that sets up the environment... any help is appreciated.. thanks for sharing 👍

You can use the VirtualBox with ubuntu linux :)