pulkin/micropython #100: No response from REPL prompt over serial after 10-20 secs

midasgossye, 2021-02-12:
Hi,

I managed to update the firmware on the A9G board to he micropython port.
After connecting a USB<->serial adapter on TX/RX1 I get the micropython prompt as expected.
I am able to type in commands and get responses for 10-20 secs, but then it hangs and doesn't respond at all until I repower the board. No errors are thrown in the serial terminal before it becomes unresponsive.

One additional note: this doesn't seem to happen when the SIM card is not inserted.

Any help or thoughts on how to solve this problem would be very welcome, since I was really enjoying the micropython port on the A9G before this issue started to happen when I updated the firmware to the latest version on this github repo...

midasgossye, 2021-02-12:
These are the events that the coolwatcher debug program shows when this happens, not sure if this is helpful but I'll include it below:
```
Detected event: 0xca1bc2c0.
Detected event: 0x9999a244.
Detected event: 0x77770006.
Detected event: 0xffffff03.
Detected event: 0xb0070000.
Detected event: 0x6c67f234.
Detected event: 0x89000000.
Detected event: 0x99009000.
Detected event: 0x89110001.
Detected event: 0x89110005.
Detected event: 0x8911116b.
Detected event: 0x89019000.
Detected event: 0x89110009.
Detected event: 0x8811556b.
Detected event: 0x89110005.
Detected event: 0x89111139.
Detected event: 0x89019000.
Detected event: 0x89110006.
Detected event: 0x89112239.
Detected event: 0x89300100.
Detected event: 0x8930064d.
Detected event: 0x89300568.
Detected event: 0x89300668.
Detected event: 0x89300542.
Detected event: 0x89300642.
Detected event: 0x89300557.
Detected event: 0x89300657.
Detected event: 0x89300544.
Detected event: 0x89300553.
Detected event: 0x89300507.
Detected event: 0x89300550.
Detected event: 0x89300548.
Detected event: 0x8930056b.
Detected event: 0x89300563.
Detected event: 0x8930053b.
Detected event: 0x89300594.
Detected event: 0x89300994.
Detected event: 0x89300597.
Detected event: 0x2244420b.
Detected event: 0x22446804.
Detected event: 0x23000042.
Detected event: 0x23000068.
Detected event: 0x22445c00.
Detected event: 0x22445d40.
Detected event: 0x23000061.
Detected event: 0x2300005d.
Detected event: 0x2300003b.
Detected event: 0x23000063.
Detected event: 0x2300008e.
Detected event: 0x23000062.
Detected event: 0x2300003f.
Detected event: 0x22445f09.
Detected event: 0x2300005f.
Detected event: 0x55000019.
Detected event: 0x1204aaaa.
Detected event: 0x00000001.
```

bokolob, 2021-02-28:
Hi. check that power button is pulled correctly. I don’t remember exactly up or down, in any case it shuts down in short time without

midasgossye, 2021-03-02:
Thanks for your suggestion. I added some extra capacitors to the power input (see [this issue](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/421)) and reflashed the firmware, and that seemed to do the trick. I'll defintely also check the power button pin to get it to run stabily.

closed 2021-03-02 by midasgossye