pulkin/micropython #109: Network status is always '3' with the sim

floor12, 2022-04-09:
Hello! 

I have a few IoT sim cards from Olivia Wireless.
The sim card works fine on stock AT firmware, but with micro python i have `is_network_registred()` is true, `get_network_status()` is 3, and it halts on trying `cellular.gprs('RH','','')`.

The firmware is 
`MicroPython v1.9.4-2810-gf467d36d0-dirty on 2020-09-12; A9/A9G module with RDA8955`

The module works fine with other sim cards of my country (Spain), but not with the IoT sim.

pulkin, 2022-04-17:
It could be #28 but I think I patched those.

floor12, 2022-04-17:
First, although the azure build should include this fix, but it does not include it. The build from azure also halts when trying to use a 2 character APN. It seems that the pipeline needs to be updated - to add the patching of a9g.

I applied your patches to the latest version of a9g sdk and rebuilt the firmware. After that, the freezes with 2-letter APN stopped. However, gprs still does not work with oliviawireless sim cards and micropython. Now it throws timeout exception.  The network is registered, but network status is always 3, sometimes 4.

It seems that the behavior of these SIM cards (may be the order of events) is a little different from the usual ones. When i use stock AT-modem firmware, on command AT+CREG? it returns 1,5 except of 1,1. But connection is works.

commit c274c947 (2022-04-19): build: fix #109
closed 2022-04-19 by pulkin