pulkin/micropython #116: \xFF is missing from UART.read()

garudaonekh, 2022-08-09:
Hi,
I send data from a LORA RS485 via UART to another. All the 0xFF byte are missing. At first, I thought it maybe the LORA Module have problem, I try another pair from a different company. Exactly same result. I try with computer UART, the ff byte is received normally.
For example:

`uart.write(b'\xff\xff\x01\x02')`
On the receiver will receive as:

`b'\x01\x02'`

The problem here is with the read(). 

Thanks;

garudaonekh, 2022-08-09:
if I put the board to sleep, all the data coming correctly into the buffer and I can read after wakeup from time.sleep(). but if data is coming while the board is running, all \xFF are missing.

robert-hh, 2022-08-10:
The culprit is the function uart_rx_intr_handler() in uart.c, which suppresses all characters that happen to be the mp_interrupt_char(). A simple fix could be:
```
static void uart_rx_intr_handler(UART_Callback_Param_t param) {
    // handles rx interrupts
    ringbuf_t *ringbuf = uart_ringbuf + param.port - 1;
    int* to_dupterm = uart_attached_to_dupterm + param.port - 1;
    for (uint32_t i=0; i<param.length; i++) {
        if (*to_dupterm && param.buf[i] == mp_interrupt_char) {
            mp_keyboard_interrupt();
        }
        ringbuf_put(ringbuf, param.buf[i]);
    }
}
```
Besides that, the data type of param.buf in the type UART_Callback_Param_t must be set to unsigned char instead of signed char. Otherwise setting mp_interrupt_char to -1 will not have the intended effect of ignoring the interrupt.

closed 2022-08-11 by garudaonekh

robert-hh, 2022-08-11:
Since this is a bug for everyone the issue better stays open to document it.