pulkin/micropython #117: Can Neo M6 module mircropython library work with this port?

Pankaj-Talesara, 2022-08-11:
I found a Neo M6 library for Arduino [here](https://microcontrollerslab.com/neo-6m-gps-module-esp32-micropython/). It is for esp32. 
Will it work with micropython ported for A9G.
I know A9G has gps inbuilt but I am having issues using it. It always gives fixed location while Neo Work flawlessly with arduino,

ChitlangeSahas, 2022-08-27:
Are you sure that you have enough satellites? The gps functions return the last fixed location, so you might not have enough satellites and hence it's just returning the last location. @Pankaj-Talesara

garudaonekh, 2022-12-07:
@Pankaj-Talesara why not? Neo M6 use UART so it'll work with any MCU with UART support. but you may need to use A9 instead of A9G as the UART port already used by A9G's GPS 

I have used A9G's GPS, it works fine but its antenna is not good so you need to put it outside otherwise it'll not fix any GPS location.