pulkin/micropython #121: Unable to detect SIM card while running on battery

techiesms, 2023-04-30:
I tried writing script that send SMS that contains GPS coordinates in it. The script works fine when I test with USB to TTL Programmer attached with the A9G Board. But when I remove that and try to power A9G board with battery or direct USB it doesn't work at all. It say's SIM not inserted 

can anyone tell me the exact problem ?

machiobrian, 2024-01-27:
Have you checked the power drawn while establishing a GSM connection?

machiobrian, 2024-01-27:
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/421

enekux, 2024-03-22:
Hi, I am having the same problem as you @techiesms. Did you ever found a solution to this problem? Thanks and regards!

enekux, 2024-05-24:
Hi @techiesms did you ever find a solution to this? Thanks