pulkin/micropython #124: How can I use nmea_data() ?

spliff1, 2023-07-11:
Hello!

I've flashed this project on my A9G board and it works great! I need to use this board as a GPS tracker for a vehicle. Tracking and reporting back to a Traccar server works great, but I also want to get the altitude and speed, and since there is no other way to get that information, I guess I have to mess with nmea.

gps.nmea_data() would return something like this:
`((742413785, False, 4127.91650390625, 2904.127197265625, 0.0, 0.0, nan), ((65, 1, bytearray(b'\x1e\x00\x00\x00'), nan, nan, nan), (65, 1, bytearray(b'\x00\x00\x00\x00'), nan, nan, nan)), (64985, 4127.91650390625, 2904.127197265625, 0, 1, nan, 102.8000030517578, 77, 36.0, 77, nan), (nan, nan, 0, 0, 0), (0, nan, nan, nan, nan, nan, nan, nan), ((3, 1, 11, ((7, 73, 205, 0), (9, 67, 58, 0), (30, 42, 220, 27), (4, 37, 79, 0))), (3, 2, 11, ((11, 31, 285, 18), (20, 28, 311, 0), (6, 28, 233, 26), (16, 21, 51, 0))), (3, 3, 11, ((3, 9, 141, 0), (5, 3, 317, 0), (26, 1, 29, 0), (0, 0, 0, 0))), (1, 1, 0, ((0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0))), (1, 1, 0, ((0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0))), (1, 1, 0, ((0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0))), (1, 1, 0, ((0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0))), (1, 1, 0, ((0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0))), (1, 1, 0, ((0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0))), (0, 0, 0, ((0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0))), (0, 0, 0, ((0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0))), (0, 0, 0, ((0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0))), (0, 0, 0, ((0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0))), (0, 0, 0, ((0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0))), (0, 0, 0, ((0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0))), (0, 0, 0, ((0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0))), (0, 0, 0, ((0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0))), (0, 0, 0, ((0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0))), (0, 0, 0, ((0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0))), (0, 0, 0, ((0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0), (0, 0, 0, 0)))), (0.0, nan, 0.0, 0.0, 78), (4292202496, 0, 0))`

How can I use that data to get what I need in a variable ? I don't know what each numbers actually represents.

pulkin, 2023-07-11:
I guess the answer is here https://github.com/pulkin/micropython/blob/c274c947c611f510fd2b1c4ef6cbd9f4283794fc/ports/gprs_a9/modgps.c#L325

spliff1, 2023-07-11:
Thank you for pointing me in the right direction. If I understood the code correctly...I need to access tuple 2 to get GGA data, and tuple 6 to get the VTG data:
```
nmea = gps.nmea_data()        
gga = nmea[2]
vtg = nmea[6]
```

However, GGA doesn't seem to be correct, or some information is missing - I can't figure out which is what:
`(77968, 4127.91748046875, 2204.12890625, 0, 1, nan, 108.8000030517578, 77, 36.0, 77, nan)`

And for the VTG I assume that index 3 or 4 is the reported speed ?
`(0.0, nan, 0.001000000047497451, 0.001000000047497451, 78)`

closed 2023-07-13 by spliff1