pulkin/micropython #127: GPS and GPRS not working at the same time

GDev-4664, 2024-01-05:
I am trying to implement a tracker device in my car. I am using A9G board to develop the tracker. **The GPS and GPRS works individually very fine. But when I try to push real gps data to mqtt over gprs gps package provides 0 tracked satellites.** Is there any problem with power or code ? Do I need to customize any hardware?
I am using XL4015 DC-DC converter. The output of XL4015 is 5V-2A

```
import machine
import time
import cellular
import gps
import ujson
from umqtt import simple
import utime

# Constants
LED_PIN = 27
CONNECT_RETRY_DELAY = 5
TOPIC = "username/feeds/mycar"
MQTT_NAME = "a9g-micropython-board"
MQTT_SERVER = "io.adafruit.com"
MQTT_PORT = 1883
MQTT_USERNAME = "username"
MQTT_PASSWORD = "pass"
SECONDS_IN_DAY = 24 * 60 * 60
SECONDS_MICROPYTHON_TO_UNIX = 946684800
led = machine.Pin(LED_PIN, machine.Pin.OUT, 1)

green_led = machine.Pin(30, machine.Pin.OUT, 0)
red_led = machine.Pin(26, machine.Pin.OUT, 0)
blue_led = machine.Pin(25, machine.Pin.OUT, 0)
isFirstTime = True
last_published_time = 0
client = simple.MQTTClient(MQTT_NAME, MQTT_SERVER, MQTT_PORT, MQTT_USERNAME, MQTT_PASSWORD)

def allOutLed():
    green_led.value(0)
    red_led.value(0)
    blue_led.value(0)

def publish_gps_data(client, satellites, location, formatted_datetime):
    data_array = [
        satellites[0],
        satellites[1],
        formatted_datetime,
        location[0],
        location[1],
    ]
    json_data = ujson.dumps(data_array)
    client.publish(topic=TOPIC, msg=json_data, retain=False, qos=1)
    led.value(1)

def get_current_time():
    return utime.time()

def connect_to_cellular():
    global isFirstTime

    while True:

        print("I am in cellular Loop")
        allOutLed()
        red_led.value(1)
        if not isFirstTime:
            if cellular.gprs():
                break
        try:
            cellular.gprs("apn_name", "", "")
            break
        except Exception as e:
            print("Error connecting to cellular:", e)
            time.sleep(CONNECT_RETRY_DELAY)
    isFirstTime = False
    return

def connect_to_mqtt():
    while True:
        allOutLed()
        blue_led.value(1)
        try:
            client = simple.MQTTClient(MQTT_NAME, MQTT_SERVER, MQTT_PORT, MQTT_USERNAME, MQTT_PASSWORD)
            client.connect()
            break
        except Exception as e:
            print(e)
            continue
    return

def main():
    allOutLed()
    while True:
        connect_to_cellular()
        try:
            connect_to_mqtt()
        except Exception as e:
            print("Error:", e)
            continue
        gps.on()
        allOutLed()
        green_led.value(1)
        publish_gps_data(client, gps.get_satellites(), gps.get_location(), gps.time())
        time.sleep(20)
        

if __name__ == "__main__":
    main()

```

pulkin, 2024-01-05:
2 amps is probably not enough

GDev-4664, 2024-01-05:
Should I Try with 3amps?

pulkin, 2024-01-05:
I think you should try turning off gps before sending the data

GDev-4664, 2024-02-05:
Not working brother. The problem is when the mqtt is connected the tracked satellite is always zero

Pr0va1der, 2024-05-19:
Have you been able to solve this problem? I came across the same one, but I'm sending over HTTP